davidthings/tinyfpga_bx_usbserial #1: Mac OS issue (endpoints?)

daviddurand, 2019-05-13:
I'm trying to make this create a serial port in Mac OS X (Mojave) and it's not being recognized. I think that there may be some dependency on the Endpoint numbers, in particular, I think it wants 3 different endpoints for ACM control, in and out. The descriptor sets the endpoints like this:
```
  `define CDC_ACM_ENDPOINT 2
  `define CDC_RX_ENDPOINT 1
  `define CDC_TX_ENDPOINT 1
```
I've tried changing this to something like this (which matches the teensy config):
```
  `define CDC_ACM_ENDPOINT 2
  `define CDC_RX_ENDPOINT 3
  `define CDC_TX_ENDPOINT 4
```
However, I'm pretty sure that this can't be enough, as I can't figure out where endpoints in the incoming and outgoing USB packets are being parsed/set.

I may be wrong about the endpoints, but some of the descriptors have shown signs of working, so I think that might be the right track.

davidthings, 2019-05-13:
I'm mostly focused on Linux...  But I'd be super surprised if Mac OS and Linux had different requirements for CDC.

Can you get your device to program under the Mac?  If so, that's interesting because the USB logic for programming is very similar to the USB logic in this library. 

There are some messages on the TinyFPGA forum where people have needed to do some minor adjustments to get the TinyFPGA to work on the Mac.  Perhaps they can help?

Let me know if you can get it working and I'll put the hints in the ReadMe.

daviddurand, 2019-05-13:
So I think the basic setup is pretty similar, and the device class drivers used to work on both platforms pretty regularly. My BX works fine for programming, but tinyprog now uses direct device control via libusb rather than a serial device. Recent Mac versions are also very picky about driver signing, making the easy hacks harder than they once were.

I've been through all the forum posts and a bunch of other posts across the net.

Based on this, and comparing to serial devices that do work on the Mac, I also believe that recent driver changes may have embedded some assumptions about what the endpoints should be, even though the descriptors should be the only relevant factor.

If I change those configs, I'm not sure if the right state machines will follow those changes. It doesn't look like data is traveling out of those descriptors, but of course I could be missing it.

My question is if there are other spots in the usb controller that embed knowledge of the endpoint numbers, and would need to be updated. This would help me  have on the serial problem, but could also help me work on midi, HID, MIDI, and perhaps even audio class devices.

I do not yet know verilog well enough to trace through the whole usb controller in a reasonable time, but I have a fingernail group on USB from a developer/protocol point of view, so I think I might be able to make some progress.

davidthings, 2019-05-14:
Maybe ask on the TinyFPGA BX discourse forum?  There are a lot of variations of this USB code floating around and I bet someone there has used it on the Mac.