davidthings/tinyfpga_bx_usbserial #9: No Device shows up on echo example

bmentink, 2020-04-08:
Hi,

I build the example "echo" code, it built without errors and programmed the device (on Linux host), however no device was presented when the image booted.

I could tell the bootloader was no longer active, as the led stopped flashing.
I also tried tinyprog -b , but same issue, no device.

I have the latest TinyFPGA-BX from cloud source ..

Any idea's of what to try next?

bmentink, 2020-04-09:
Found this message from the driver:

[1988064.213076] cdc_acm 1-4.3:1.0: failed to set dtr/rts
[1988064.307113] usb 1-4.3: USB disconnect, device number 35
[1988064.521762] usb 1-4.3: new full-speed USB device number 36 using ehci-pci
[1988064.601819] usb 1-4.3: device descriptor read/64, error -32
[1988064.788437] usb 1-4.3: device descriptor read/64, error -32
[1988064.968423] usb 1-4.3: new full-speed USB device number 37 using ehci-pci
[1988065.048419] usb 1-4.3: device descriptor read/64, error -32
[1988065.235059] usb 1-4.3: device descriptor read/64, error -32
[1988065.342165] usb 1-4-port3: attempt power cycle
[1988065.935139] usb 1-4.3: new full-speed USB device number 38 using ehci-pci
[1988066.348484] usb 1-4.3: device not accepting address 38, error -32
[1988066.421734] usb 1-4.3: new full-speed USB device number 39 using ehci-pci
[1988066.835127] usb 1-4.3: device not accepting address 39, error -32
[1988066.835538] usb 1-4-port3: unable to enumerate USB device

nalzok, 2020-04-10:
If you are on macOS, try applying this patch: https://github.com/davidthings/tinyfpga_bx_usbserial/pull/2/commits/42b8902d5191308ead43f18c3c9b1cd16594f4ba

bmentink, 2020-04-10:
I am on Linux (Arch), does it still apply?

nalzok, 2020-04-10:
I’m not sure. At least that patch fixed my issue on macOS. It doesn’t hurt to try, I suppose?

bmentink, 2020-04-11:
No help, same error after the patch .. have you tried on Linux?

nalzok, 2020-04-11:
Nope, unfortunately, I don't have a Linux box at my hand :(

bmentink, 2020-04-11:
Any other suggestions on how to debug this?

nalzok, 2020-04-11:
Maybe you can check out the [official TinyFPGA forum](https://discourse.tinyfpga.com/). Particularly [this thread](https://discourse.tinyfpga.com/t/usb-communication/613/58) might interest you.

bmentink, 2020-04-11:
The examples from the TinyFPGA site work fine .. it is this code that does not work. But, I will have a look

bmentink, 2020-04-12:
Quick question, did you use "make" to build of "apio build"

nalzok, 2020-04-12:
I used `make prog`.

bmentink, 2020-04-12:
I tried to compare the files (endpoint)  that you linked  with your endpoint code, they are totally different, so the advice in that thread does not help ..

Could you possible install linux in a VM on your Apple and do some testing? I really need your code working, as I have a Forth CPU in my fpga code that needs a serial console to upload code on the fly ..

I don't really want to add an FTDI USB->Serial device to do this ..  (as is on the Lattice dev board)

nalzok, 2020-04-13:
> I tried to compare the files (endpoint) that you linked with your endpoint code, they are totally different, so the advice in that thread does not help ..

Which linked files are you referring to, issue #8 or something else?

By the way, I did a quick search on Google and found [this issue](https://github.com/tinyfpga/TinyFPGA-BX/issues/4), which resembles your problem.

Sorry, but I don't have much time for this recently. Good luck! (Also, I am not the author; just a random user of this repository :)

bmentink, 2020-04-13:
Yep, I had found that issue on google too, but does not relate to my issue. That issue realted to permissions for the tty device, I don't even have a tty device showing up, as I originally stated.

I nned to know why the enumeration is failing ... I don't even know what error code -32 means, maybe you do?

Sorry you don't have time, maybe you are not in lock-down like we are in NZ

nalzok, 2020-04-13:
Unfortunately, I have no clue what error -32 means. You might be able to find something useful on Google or Stack Overflow. Also, I have heard that the Arch Linux community is very helpful, so hopefully, you will find some hint by asking around on their forum (), mailing list, IRC channel, etc, while waiting for the author @davidthings's insight.

With enough time and determination, you can also look into the driver source to investigate it yourself, but yeah, that would take a lot of expertise and time. I suppose you will learn a lot in this process, though.

By the way, have you tried using another Linux box? Maybe, just maybe, there is something wrong with your workstation, rather than the FPGA device.

I am currently in Canada, and we are also in self-isolation, but I still need to work remotely.

bmentink, 2020-04-14:
Hi, it is nothing to do with my Linux box, I can see the TinyFPGA Bootloader serial port just fine ... the problem is with this code ..

i.e I can load a program with the bootloader and run it, eg some of the example codes 

I wish the author would chime in ...

bmentink, 2020-04-16:
I have it working. The demo echo code still does not work for me, but I included the code into a current project, and it echoes characters just fine.

No I need to work out how to interface it with my other code that was working with a standard uart.. Are there any examples out there of using this pipeline interface?

nalzok, 2020-04-16:
That's awesome!

As for the examples, I have been pulling my hair out trying to understand the pipeline interface for a while, and _just_ got it to work a few days ago (see #8). Here are two trivial examples that you might find helpful: [usbserial_examples.zip](https://github.com/davidthings/tinyfpga_bx_usbserial/files/4485026/usbserial_examples.zip)

+ `usbserial_hello.v` prints "hello, world" to a serial console.
+ `usbserial_pipeline.v` creates a pipelined loopback.

The module `usb_uart` seems to expect the signal `uart_in_data` to be asserted for one more cycle after the transmission is considered complete, so I added the `pardon` register as a workaround. I suppose this bug/feature goes back into [tiny_usb_examples](https://github.com/lawrie/tiny_usb_examples) (which by the way also includes a number of examples as discussed on the [forum thread](https://discourse.tinyfpga.com/t/usb-communication/613) by @lawrie.griffiths).

bmentink, 2020-04-16:
Many Thanks,

Your code will come in handy. I am trying to emulate my current memory mapped uart interface which is defined as:

  usbuart  usbuart_ (
     .clk(clk),
     .reset(reset),
     .rd(uart0_rd),
     .wr(uart0_wr),
     .valid(uart0_valid),
     .busy(uart0_busy),
     .tx_data(dout_[7:0]),
     .rx_data(uart0_data)
  );

I will see how I get on ..

nalzok, 2020-04-16:
Great! Could you post your code once you are done? I was trying to integrate this repository to PicoSOC, as discussed on [this forum thread](https://discourse.tinyfpga.com/t/how-to-make-picosoc-speak-usb-on-tinyfpga-bx/1358). I have a lot of stuff to do recently so I will not work on this for a while, but your code will be a great reference to me in the future.

bmentink, 2020-04-17:
Hi,
I have it partially working. The sending from my terminal to my cpu works. It is the sending from the cpu to the terminal that has similar issues that you had with multiple characters send for everyone received. This is what  I have so far, I don't know how to incorporate the "busy" flag or the "wr" flag yet  .... currentl if I send it a character it expects, I get that character plus a bunch of un-wanted ones back ..

-------------------------------------------------
module usbuart (
    input               clk,        // The master clock for this module
    input               reset,     // Synchronous reset, active high
    input               rd,         // read strobe  -- used only to clear valid flag.
    input               wr,         // write strobe   
    inout               pin_usb_p,
    inout               pin_usb_n,
    output              pin_pu,
    output reg          valid,      // Indicates a new byte is available. clears on read.
    output reg          busy,       // Low when transmit line is idle.
    input [7:0]         tx_data,    // Byte to transmit
    output reg [7:0]    rx_data    // *Most recent* byte received -- whether or not the last was collected.
    );

    // uart pipeline in
    wire [7:0] uart_in_data;
    wire       uart_in_valid;
    wire       uart_in_ready;

    // uart pipeline out
    wire [7:0] uart_out_data;
    wire       uart_out_valid;
    wire       uart_out_ready;

    usb_uart uart (
        .clk_48mhz  (clk),
        .reset      (reset),

        // pins
        .pin_usb_p( pin_usb_p ),
        .pin_usb_n( pin_usb_n ),

        // uart pipeline in
        .uart_in_data( uart_in_data ),      // in
        .uart_in_valid( uart_in_valid ),    // in
        .uart_in_ready( uart_in_ready ),    // out

        // uart pipeline out
        .uart_out_data( uart_out_data ),    // out
        .uart_out_valid( uart_out_valid ),  // out
        .uart_out_ready( uart_out_ready  )  // in

    );

    reg pardon;

    //  A sender will place new data on the data port and raise the valid flag. 
    // If a receiver is ready to receive, it will raise the ready flag

    always @(posedge clk) begin
        if (reset) begin
            uart_out_ready <= 1;
            busy <= 0;
            pardon <= 0;
            valid  <=0;
        end else begin

            // Sending to terminal  @@@@@ this needs work , currently only getting one character then stops @@@@
            if (pardon) begin
                pardon <= 0;
                uart_in_valid <= 0;
            end else if (uart_in_valid ) begin
                    if (!uart_in_ready) begin
                        pardon <= 1;
                    end
            end else begin  
                    uart_in_valid <= 1; 
                    uart_in_data <= tx_data & 8'h7f;   
            end 

            // Receiving from terminal  @@@@@@ THIS works !!!!!! @@@@@
            if (uart_out_valid ) begin
                rx_data <= uart_out_data;   // Fetch data byte
                valid <= 1;
            end 
            // reset valid flag, and let pipline know we are ready for another
            if(rd && valid) begin 
                uart_out_ready <= 1;
                valid <= 0;
            end
            

        end
    end

    // USB Host Detect Pull Up
    assign pin_pu = 1'b1;
endmodule

bmentink, 2020-04-21:
Any suggestions? I, like you, am pulling my hair out ...

bmentink, 2020-04-21:
The following code sort of works. I can interact with the CPU, the only problem is that there is a one character delay on echo'ing characters .. i.e if I type 1234 I see 234, it is always one character behind. Maybe you can see the problem?

```
// USB UART
//
module usbuart (
    input               clk,        // The master clock for this module
    input               reset,     // Synchronous reset, active high
    input               rd,         // read strobe  -- used only to clear valid flag.  
    input               wr,
    inout               pin_usb_p,
    inout               pin_usb_n,
    output              pin_pu,
    output          valid,      // Indicates a new byte is available. clears on read.
    output           busy,       // Low when transmit line is idle.
    input [7:0]         tx_data,    // Byte to transmit
    output reg [7:0]    rx_data    // *Most recent* byte received -- whether or not the last was collected.
    );

    // uart pipeline in
    wire [7:0] uart_in_data;
    wire       uart_in_valid;
    wire       uart_in_ready;

    // uart pipeline out
    wire [7:0] uart_out_data;
    wire       uart_out_valid;
    wire       uart_out_ready;

    usb_uart uart (
        .clk_48mhz  (clk),
        .reset      (reset),

        // pins
        .pin_usb_p( pin_usb_p ),
        .pin_usb_n( pin_usb_n ),

        // uart pipeline in
        .uart_in_data( uart_in_data ),      // in
        .uart_in_valid( uart_in_valid ),    // in
        .uart_in_ready( uart_in_ready ),    // out

        // uart pipeline out
        .uart_out_data( uart_out_data ),    // out
        .uart_out_valid( uart_out_valid ),  // out
        .uart_out_ready( uart_out_ready  )  // in

    );

    reg busy_;
    reg [3:0] state;

    //  A sender will place new data on the data port and raise the valid flag. 
    // If a receiver is ready to receive, it will raise the ready flag
    always @(posedge clk) begin
        if (reset) begin
            uart_out_ready <= 1;
            uart_in_valid <= 0;
            busy_ <= 0;
            valid  <=0;
            state <= 1;
        end else begin

            // Sending to terminal           
            case (state) 
                0:  if(uart_in_ready) begin
                        uart_in_valid <= 0;
                        busy_ <= 0;
                        state <= 1;
                    end
                1:  if(wr) begin   
                        uart_in_data <= tx_data;
                        uart_in_valid <= 1; 
                        busy_ <= 1;
                        state <= 2; 
                    end
                2:  begin 
                        state <= 0;
                    end
            endcase

           
            // Receiving from terminal  
            if (uart_out_valid ) begin
                rx_data <= uart_out_data;   // Fetch data byte
                valid <= 1;
            end 
            // reset valid flag, and let pipeline know we are ready for another
            if(rd && valid) begin 
                uart_out_ready <= 1;
                valid <= 0;
            end
            

        end
    end
    assign busy = busy_;
    // USB Host Detect Pull Up
    assign pin_pu = 1'b1;
endmodule
```

nalzok, 2020-04-23:
Hi, I haven't got a lot of time, but [someone](https://github.com/davidthings/tinyfpga_bx_usbserial/issues/8#issuecomment-618679076) just told me that the `uart_out_ready` signal is active **high** instead of active low as I have assumed before. Since you adopted my code, I suppose you are probably also under the wrong impression.

Hope this clarification helps!

bmentink, 2020-04-23:
I had assumed it was active high, but thanks anyway ...

bmentink, 2020-04-26:
I have it all working. 

Some of my issue has been with emulating a uart as for instance, if I press carriage return on my terminal, it only sends \r (13) it does not send LF (10) as well, which my software expects ..

The code I have now is: 

```
// USB UART
//
module usbuart (
    input               clk,        // The master clock for this module
    input               reset,     // Synchronous reset, active high
    input               rd,         // read strobe  -- used only to clear valid flag.  
    input               wr,
    inout               pin_usb_p,
    inout               pin_usb_n,
    output              pin_pu,
    output              valid,      // Indicates a new byte is available. clears on read.
    output              busy,       // Low when transmit line is idle.
    input [7:0]         tx_data,    // Byte to transmit
    output reg [7:0]    rx_data    // *Most recent* byte received -- whether or not the last was collected.
    );

    // uart pipeline in
    wire [7:0] uart_in_data;
    wire       uart_in_valid;
    wire       uart_in_ready;

    // uart pipeline out
    wire [7:0] uart_out_data;
    wire       uart_out_valid;
    wire       uart_out_ready;

    usb_uart uart (
        .clk_48mhz  (clk),
        .reset      (reset),

        // pins
        .pin_usb_p( pin_usb_p ),
        .pin_usb_n( pin_usb_n ),

        // uart pipeline in
        .uart_in_data( uart_in_data ),      // in
        .uart_in_valid( uart_in_valid ),    // in
        .uart_in_ready( uart_in_ready ),    // out

        // uart pipeline out
        .uart_out_data( uart_out_data ),    // out
        .uart_out_valid( uart_out_valid ),  // out
        .uart_out_ready( uart_out_ready  )  // in

    );

    reg busy_, valid_;
    reg uart_in_valid_, uart_out_ready_;
    reg [7:0] uart_in_data_;
    reg [3:0] state;
    reg [3:0] state_rx;

    
    always @(posedge clk) begin
        if (reset) begin
            uart_out_ready_ <= 1;
            uart_in_valid_ <= 0;
            busy_ <= 0;
            valid_  <=0;
            state <= 0;
            state_rx <= 1;
        end else begin

            // Sending to terminal       
            //  A sender will place new data on the data port and raise the valid flag.    
            case (state) 
                0:  if(wr) begin   
                        uart_in_data_ <= tx_data;
                        uart_in_valid_ <= 1;    // We have data raise the valid flag
                        busy_ <= 1;
                        state <= 1; 
                    end
                1:  if(uart_in_ready) begin     // Check pipline is ready
                        uart_in_valid_ <= 0;
                        busy_ <= 0;
                        state <= 0;
                    end
            endcase

            // Receiving from terminal      
             // If a receiver is ready to receive, it will raise the ready flag
            case (state_rx) 
                0:  if(rd && valid_) begin      // use rd to reset valid flag
                        uart_out_ready_ <= 1;   // Raise ready flag, we have read last byte, ready for new ..
                        valid_ <= 0;
                        state_rx <= 1;
                    end
                1:  if(uart_out_valid) begin    // If we have a character
                        rx_data <= uart_out_data;
                        uart_out_ready_ <= 0;   // reset ready flag..
                        valid_ <= 1;
                        state_rx <= 0;
                    end
            endcase 

            
        end
    end

    assign busy = busy_;
    assign valid = valid_;
    assign uart_in_valid = uart_in_valid_;
    assign uart_out_ready = uart_out_ready_;
    assign uart_in_data = uart_in_data_;

    // USB Host Detect Pull Up
    assign pin_pu = 1'b1;
endmodule

```

bmentink, 2020-04-26:
The code is "very" fast, way faster than a standard UART, even when the hardware UART runs at 1Mbaud ... my hat off to you David ..

nalzok, 2020-04-27:
@bmentink Congratulations! Things are tricky in FPGA, even if all you need to implement is a different handshake protocol instead of something fancier :)

Also, last time I worked with this repository, a HUGE problem is timing (and it still haven't been resolved). Since it requires a 48MHz clock while the rest of my design runs at 24MHz, I needed to somehow cross different clock domains, and there come metastability issues. Essentially I need something like

    usb_uart_arbitrary_clk uart (
        .clk_whatever_mhz  (clk_whatever_mhz),
        .reset             (reset),

        // pins
        .pin_usb_p         (pin_usb_p),
        .pin_usb_n         (pin_usb_n),

        // uart pipeline in
        .uart_in_data      (uart_in_data),
        .uart_in_valid     (uart_in_valid),
        .uart_in_ready     (uart_in_ready),

        // uart pipeline out
        .uart_out_data     (uart_out_data),
        .uart_out_valid    (uart_out_valid),
        .uart_out_ready    (uart_out_ready)
    );

where all the `uart_*` signals sync with `clk_whatever_mhz` and the wrapper module will automatically instantiate a PLL and derive a 48MHz clock. It is said that I need a FIFO to avoid metastability, but I haven't figured it out yet.

If the rest of your design doesn't happen to run at 48HMz, maybe you'll have some advice for me?

bmentink, 2020-04-27:
Hi,

My design all runs at 48Mhz, so I don't have your problem.

However, my understanding is that you can still run the USB-UART at 48Mhz and apply some code to the interconnecting lines (i.e pipeline in and out)  there is plenty of documentation out there for syncing control lines to a different clock domain ... just google around ..

Search on crossing clock domains regarding control lines .... the steps are:

1. run your pll at 48Mhz
2. derive your 24Mhz off the above 48Mhz
3. run the pipeline signals through "some logic" that sync's them across clock domains ..

Somewhere to start would be here:
https://www.fpga4fun.com/CrossClockDomain.html

nalzok, 2020-04-27:
Thanks for the advice. I'll definitely look into that!

bmentink, 2020-05-02:
@naizok

This may be what you are looking for:   (from the bootloader project ..)
https://github.com/tinyfpga/TinyFPGA-Bootloader/blob/master/common/strobe.v

bl0x, 2020-11-28:
> I have it working. The demo echo code still does not work for me, but I included the code into a current project, with my own Makefile and it echoes characters just fine.
> 
> Now I need to work out how to interface it with my other code that was working with a standard uart.. Are there any examples out there of using this pipeline interface?

@bmentink 

I'm experiencing a similar issue: I'm using the code as-is from this repo, run it through an up to date toolchain:

* Yosys 0.9+3675 (git sha1 71ca9a82, gcc 10.2.0 -march=native -mtune=native -O2 -fstack-protector-strong -fno-plt -fPIC -Os)
* nextpnr-ice40 -- Next Generation Place and Route (Version cf331ca2)

The resulting binary is uploaded to the TinyFPGA BX and what happens then differs between Arch Linux and Windows:

On Arch Linux I'm getting this:

```
[237334.779127] usb 3-1.2: new full-speed USB device number 18 using ehci-pci
[237334.877663] usb 3-1.2: New USB device found, idVendor=1d50, idProduct=6130, bcdDevice= 0.00
[237334.877668] usb 3-1.2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[237334.878175] cdc_acm 3-1.2:1.0: ttyACM0: USB ACM device
[237338.635742] usb 3-1.2: reset full-speed USB device number 18 using ehci-pci
[237338.736137] usb 3-1.2: device firmware changed
[237338.736508] usb 3-1.2: USB disconnect, device number 18
[237338.955753] usb 3-1.2: new full-speed USB device number 19 using ehci-pci
[237339.029035] usb 3-1.2: device descriptor read/64, error -32
[237339.209059] usb 3-1.2: device descriptor read/64, error -32
[237339.389066] usb 3-1.2: new full-speed USB device number 20 using ehci-pci
[237339.462352] usb 3-1.2: device descriptor read/64, error -32
[237339.642386] usb 3-1.2: device descriptor read/64, error -32
[237339.749158] usb 3-1-port2: attempt power cycle
[237340.342370] usb 3-1.2: new full-speed USB device number 21 using ehci-pci
[237340.755693] usb 3-1.2: device not accepting address 21, error -32
[237340.829017] usb 3-1.2: new full-speed USB device number 22 using ehci-pci
[237341.242344] usb 3-1.2: device not accepting address 22, error -32
[237341.242532] usb 3-1-port2: unable to enumerate USB device
```

First the bootloader presents its /dev/ttyACM0 (which works fine together with tinyprog), then after the bootloader loads the user code, Arch Linux fails to enumerate the device. I've tried with different USB ports and even an old hub in between. Same issue.

However, if I connect to a Windows machine, everything works as expected. I can connect via a terminal emulator and I get the characters echoed back.

My question now is: Can you, @bmentink, please provide information about the differences between your Makefile and the one supplied in this example? You seem to have no trouble at all anymore running this code inside your project(s). It would be very helpful to see how you've done it.

Many thanks!

bkettle, 2022-09-30:
Sorry to revive a super dead thread, but @bl0x or a @bmentink do either of you have more information about this? I am having the same issue, also on Arch (don't have a windows machine to test with).

bl0x, 2022-09-30:
@bkettle I'm afraid not. I didn't manage to get it running and I also haven't gotten any reply from @bmentink on the matter. My solution was to use an external UART module and hook it up to two pins.

ulixxe, 2022-10-01:
Hi, maybe you can give my code a try (https://github.com/ulixxe/usb_cdc).
I didn't understand your issue, but my USB implementation allows a lower clock frequency for your design.