davidthings/tinyfpga_bx_usbserial #10: Strange behavior in Windows.

juanmard, 2020-04-14:
I am testing with a _TinyFPGA-BX_ and I get strange behavior regarding the original bootloader.
In the current configuration I can access the bootloader (COM6). It reacts correctly by sending you byte (0x00) and loads the user _bitstream_.
In device manager everything is correct (in Spanish... :-)).

[image]

Loading the new "tinyfpga_bx_usbserial" stops accessing the port and _Windows 10_ reacts with a warning (also happens with the "hello_world" of @nalzok) giving an error 0x0A.

[image]

I have tested it on a GNU/Linux distribution (Debian) and everything works perfectly however.
I suppose it is because of the CDC-ACM driver, but I don't understand what difference there is so that with the _bootloader_ it works and with this other _bitstream_ not. Any clues?

nevercast, 2020-06-18:
I believe I've resolved it, see chain here: https://twitter.com/NvrCast/status/1273614583211520004
and solution posted here: https://discourse.tinyfpga.com/t/usb-communication/613/65

The issue is that when wLength exceeds 7 bits, it wraps over and usbserial returns less bytes than it should 

https://github.com/davidthings/tinyfpga_bx_usbserial/blob/master/usb/usb_serial_ctrl_ep.v#L106-L108

This can happen because Windows (WHY!?) requests a very large Descriptor size, 265 bytes exactly. 
[image]

nevercast, 2020-06-19:
A fix for this has been merged, are you able to check if the problem is resolved for you @juanmard ?

juanmard, 2020-06-19:
Yeeeessss!
Now it's ok!

[image]

No error and example "hello_world" running... Thanks! ;)

nevercast, 2020-06-20:
Fantastic! Glad to see this is fixed for you!

closed 2020-06-22 by davidthings