davidthings/tinyfpga_bx_usbserial #14: Missing Copyright License

oskirby, 2020-06-21:
This is perhaps a minor nit to pick, but I wasn't able to find a software license for this project.

I created a fork of your project to develop into a USB/DFU bootloader (https://github.com/oskirby/tinydfu-bootloader/) and I'd like to make it open source, but I think that technically requires permission from the upstream project to create a derivative work in the absence of an open source license.

Note that the original TinyFPGA bootloader was licensed under the Apache software license.

tinyfpga, 2020-06-21:
Yeah, this needs to be released under the apache 2.0 license:

https://github.com/tinyfpga/TinyFPGA-Bootloader/blob/master/LICENSE

davidthings, 2020-06-21:
Will do.

closed 2020-06-21 by davidthings