davidthings/tinyfpga_bx_usbserial #15: Pipeline interface

keesj, 2020-06-30:
Hello, 

I have been reading the code and been trying to make more elaborate examples. Currently do not understand how 
to properly use the pipeline interface (or there is a bug). 

I would like to get the hello_world example working again and Having read https://zipcpu.com/blog/2017/08/14/strategies-for-pipelining.html
 At present it is not clear to me how the pipeline interface is supposed to work.

A few things I noticed
* uart_in_ready is only valid when the state of the pipeline is CycleData
* when setting uart_in_valid I do not know the state of the pipeline. it looks like I will need to keep uart_in_valid high for an unknown amount of time. 

Would it be possible to explain how the interfaces is  supposed to be used?
While the example (uart/echo) is working I am wondering why 

A good start for me would be a quick review of this code
https://github.com/keesj/tiny_usb_examples/blob/keesj/hello_world/src/top.v
 but explanations would be greatly appreciated!

davidthings, 2020-06-30:
Ready Valid is at once incredibly simple and a total head-scratcher.   What could be simpler, than "offer" and "acceptance" or "give" and "take".  Things change hands when both signals are present at the same time.  And yet, for the beginner (and me) when thinking in terms of FPGA it can get really complex when thinking about what is being asserted and when.  

Take a look at this Pipeline tutorial `Design` section for a little more detail. 

https://davidthings.github.io/spokefpga/pipelines#design

As is noted in the Appendix there you are actually attempting the hardest kind of pipeline code - the middleman.

My suggestion is to start by separating the incoming pipe from the outgoing one.  Maybe set the pipe in ready signal to true, and letting the in valid blink an LED.  Then you know when you're getting an incoming char.  On the outgoing side, use a timer (one of the bits of a counter) to trigger a sequence that sets the pipe out valid and loads pipe out data with the next character.  When the pipe out ready goes high, you've had a transfer and you can go to the next char or finish.

What you're attempting is a head melting place to start.  Start simple.  

It should be noted that tying the "in valid" signal to the trigger will next get you what I think you're trying to do.

Keep us posted with how you do.

keesj, 2020-07-01:
Thank you very much for the info. it is much easier to implement when you know the design behind. I got the hello world working and will continue my endeavors.

closed 2020-07-01 by keesj

keesj, 2020-07-07:
From debugging it appears to me that uart_in_ready will only become 1 uart_in_valid is first set to 1.
my guess at this point is that if uart_in_valid is not set no data out is needed and the ep does not become ready. is the interface in need of an enable signal? 

and does the following code look sane?
https://github.com/keesj/tiny_usb_examples/blob/keesj/hello_world/src/top.v#L118

davidthings, 2020-07-07:
Will check the code later, but in the mean time, yes... the uart_in_valid signal is used to signal to the usb system that there is data to send.  The usb system then creates the request to the host and allocates a buffer.  No characters are dropped and the contract of the ready-valid pair is honored.  However, all the ready valid code in here was written by a (then!) Verilog noobie.  Now I would do it differently (better, I hope).  I would signal ready, then when a character is valid, I'd consume and buffer it, and drop the ready signal until all the buffers were ready etc.

davidthings, 2020-07-07:
For your peace of mind, I will offer that I have used this code in perhaps 20 or more miniprojects including several that are aimed purely at getting 100% performance out of the interface.  It does work!