davidthings/tinyfpga_bx_usbserial #18: DRC error for Spartan7 with Vivado 2021

EmbeddedFreaks, 2021-09-16:
Hi,
I am a newbie in the world of fpga's and playing with SPARTAN Edge Accelerator Board. Trying out your code for standalone usb serial in spartan 7 but stuck at DRC error.
[DRC NSTD-1] Unspecified I/O Standard: 39 out of 39 logical ports use I/O standard (IOSTANDARD) value 'DEFAULT', instead of a user assigned specific value. This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all I/O standards. This design will fail to generate a bitstream unless all logical ports have a user specified I/O standard value defined. To allow bitstream creation with unspecified I/O standard values (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks NSTD-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run. Problem ports: debug[11:0], uart_in_data[7:0], uart_out_data[7:0], clk_48mhz, reset, uart_in_ready, uart_in_valid, uart_out_ready, uart_out_valid, usb_n_rx, usb_n_tx, usb_p_rx, usb_p_tx, and usb_tx_en.

briansune, 2022-01-08:
Hi all,

First i would like to appreciate the great work from "David Williams".
Second with ZYNQ 7010 i have no issue no the RTL implementation and testing loop-back.
So i guess it has to do with folder placement of the OS and the XDC settings.

[image]

```
module top_wrapper(
	
	input	sys_clk,
	
	// USB pins
	inout  pin_usb_p,
	inout  pin_usb_n,
	
	output     [2 : 0]     led
);
	
	wire glb_clk;
	wire glb_nrst;
	wire glb_rst;
	
	wire	[7 : 0]		uart_data;
	wire				uart_fb_rdy;
	wire				uart_fb_val;
	
	assign glb_rst = ~glb_nrst;
	assign led = {3{glb_nrst}};
	
	sys_clk_mmcm clk_gbl(
		.clk_in1	(sys_clk),
		.clk_out1	(glb_clk),
		.locked		(glb_nrst)
	);
	
	usb_uart usb_uart_inst0(
        .clk_48mhz			(glb_clk),
        .reset				(glb_rst),
		
        .pin_usb_p			(pin_usb_p),
        .pin_usb_n			(pin_usb_n),

        // uart pipeline in (out of the device, into the host)
        .uart_in_data		(uart_data),
        .uart_in_valid		(uart_fb_val),
        .uart_in_ready		(uart_fb_rdy),

        // uart pipeline out (into the device, out of the host)
        .uart_out_data		(uart_data),
        .uart_out_valid		(uart_fb_val),
        .uart_out_ready		(uart_fb_rdy),

		.debug				()
    );
	
	
endmodule
```

briansune, 2022-01-10:
I have also tested with Spartan 7.
No problem found with hot plug reset as well.
[image]